raspberrypi/pico-extras #8: Add i2s audio input

davisnando, 2021-02-23:
In the readme it's mentioned that i2s is only for audio output. Is it possible to add i2s audio input or does it maybe already exist?

kilograham, 2021-03-01:
it should be possible; doesn't currently exist

papafelice, 2021-12-22:
Are you working on this? I am in the process of attaching a real audio codec chip to the pico.
There is the usb_sound_card example that forwards the audio playback data to IO pins.
However,  the codec chips need a 250xFs clock (i.e. 12MHz when in USB mode) to work. I changed the audio_i2s.pio module to have two more SM: one to generate the 12MHz clock and one to read in the ADC data synchronous to the DAC data.

The next step would be to update audio_i2s.c to have another DMA channel for the microphone data input. 
Will this then have to be fed to the pico_audio library? Also, the usb_sound_card example, is not using the new tinyUSB stack, so my plan is to combine the low level I2S functions with the uac2_headset example.

Would this be the right approach for a real funtional USB sound card?
tinyUSB -> pico_audio -> audio_i2s -> DMAout -> PIO -> codec speaker output
tinyUSB <- pico_audio <- audio_i2s <- DMAin <- PIO <- codec mic input

Any suggestions?

lurch, 2021-12-22:
I know very little about audio, but does https://github.com/ArmDeveloperEcosystem/microphone-library-for-pico help at all? :shrug: 
(if it doesn't help, please ignore me :wink: )

papafelice, 2021-12-22:
This is using PDM (pulse density modulation), I had already checked that. Thanks anyway.

giampiero7, 2022-02-22:
Hi,
I'm gonna post this conversation here too:
https://github.com/earlephilhower/arduino-pico/issues/99

@kilograham it would be nice to have your feedback :)

Thanks!

kilograham, 2022-02-22:
oh, yes, go ahead and work on it!

biemster, 2022-11-30:
So I started with the go ahead, it's living here:
https://github.com/biemster/pico-I2S
and it's basically a copy and C++ to C port of the pico-arduino code:
https://github.com/earlephilhower/arduino-pico/tree/master/libraries/I2S/src
and has an example project using this code for an INMP441 MEMS mic here:
https://github.com/biemster/pico-INMP441

The thing is, it's not working yet :(. Most likely the AudioRingBuffer is not ported correctly, specifically the IRQ and DMA stuff I'm not sure about. I'll update here if I get it working, in the mean time PRs more than welcome!

biemster, 2022-12-01:
Hm yeah that was a short adventure, I've solved my issue by reusing the MicroPython code just like @giampiero7 did: https://github.com/biemster/pico-serialmic

jensolsson, 2024-03-23:
Any update on this, audio input would be awesome.

@biemster interesting in your reusing of the micropython code, did you switch over you project to micropython or did you use part of the micropython c code?

biemster, 2024-03-24:
@jensolsson my pico mic project got snowed under a bit, I never switched my project to micropython because I wanted to use the bitbanged ethernet stuff. It seems that I just copied the i2s code and continued as a c-sdk project (although I never finished the whole thing)