raspberrypi/pico-extras #10: CMake 3.12 or higher is required.  You are running version 3.7.2

tge96, 2021-02-24:
Hi,

Does the latest commit fix this issue?

Adding PICO_EXTRAS_PATH to ~/.bashrc
Cloning https://github.com/raspberrypi/pico-playground.git
Cloning into 'pico-playground'...
remote: Enumerating objects: 196, done.
remote: Counting objects: 100% (196/196), done.
remote: Compressing objects: 100% (144/144), done.
remote: Total 196 (delta 60), reused 174 (delta 46), pack-reused 0
Receiving objects: 100% (196/196), 1.87 MiB | 3.07 MiB/s, done.
Resolving deltas: 100% (60/60), done.
Adding PICO_PLAYGROUND_PATH to ~/.bashrc
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.12 or higher is required.  You are running version 3.7.2

-- Configuring incomplete, errors occurred!

kilograham, 2021-02-24:
no; you require a newer version of cmake to build.

tge96, 2021-02-24:
I'm building CMake 3.12 now. Thank you!

kilograham, 2021-02-24:
i guess I should have said you actually probably need 3.13!

tge96, 2021-02-24:
ok, thank you!  I install CMake 3.12.4 and re-ran pico_setup.sh and it got a little further but is now saying that I need CMake 3.13 or higher?  Why is this not installing properly on a fully updated Raspberry Pi 3B+?

Adding PICO_EXTRAS_PATH to ~/.bashrc
Cloning https://github.com/raspberrypi/pico-playground.git
Cloning into 'pico-playground'...
remote: Enumerating objects: 196, done.
remote: Counting objects: 100% (196/196), done.
remote: Compressing objects: 100% (144/144), done.
remote: Total 196 (delta 60), reused 174 (delta 46), pack-reused 0
Receiving objects: 100% (196/196), 1.87 MiB | 0 bytes/s, done.
Resolving deltas: 100% (60/60), done.
Adding PICO_PLAYGROUND_PATH to ~/.bashrc
Using PICO_SDK_PATH from environment ('/home/pi/pico/pico-sdk')
PICO_SDK_PATH is /home/pi/pico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
PICO_GCC_TRIPLE defaulted to arm-none-eabi
-- The C compiler identification is GNU 5.4.1
-- The CXX compiler identification is GNU 5.4.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
CMake Error at /home/pi/pico/pico-sdk/CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.13 or higher is required.  You are running version 3.12.4

-- Configuring incomplete, errors occurred!
See also "/home/pi/pico/pico-examples/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/pico/pico-examples/build/CMakeFiles/CMakeError.log".

tge96, 2021-02-24:
For some reason, I just now got your previous post... ok.  thank you!  I will do install 3.13 now.  is this something that pico_setup.sh should be handling? or is this the expected norm?

lurch, 2021-02-24:
`pico_setup.sh` should be installing the latest version of `cmake` installed in the APT repositories. Which is cmake 3.13.4-1 for current versions of Raspberry Pi OS.

What OS are you running on your Pi? What does `apt policy cmake` say?

> -- The C compiler identification is GNU 5.4.1

That looks a bit old too...

tge96, 2021-02-24:
Hi, this is on Raspberry Pi 3B+ built with the latest Raspian OS (recommended), as of November 2020.  The unit is up to date, as far as sudo apt update/upgrade is concerned.  I manually installed CMake 3.13.5 and pico_setup.sh got further, but is stuck on another section of the SDK install.  I posted that issue to the pico-sdk repo.

tge96, 2021-02-24:
pi@raspberrypi:~ $ sudo apt policy cmake
cmake:
  Installed: 3.7.2-1
  Candidate: 3.7.2-1
  Version table:
 *** 3.7.2-1 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status

lurch, 2021-02-24:
> 500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages

Ahhh, there's your problem. Current versions of [Raspberry Pi OS](https://www.raspberrypi.org/software/) are based on `buster`, not `stretch`. And the [release-notes](https://downloads.raspberrypi.org/raspios_full_armhf/release_notes.txt) suggest that that's been the case since June 2019!

tge96, 2021-02-24:
copy that!  is there an easy way to upgrade from stretch to buster without completely redoing the SDcard image?

lurch, 2021-02-24:
> is there an easy way to upgrade from stretch to buster without completely redoing the SDcard image?

You could _try_ following the instructions at https://www.raspberrypi.org/blog/buster-the-new-version-of-raspbian/ but there have _also_ been **many** other updates to Raspberry Pi's version of Buster since then, so you'll more than likely end up with a broken system. Much easier to just start afresh with a new SD card.

tge96, 2021-02-25:
copy that!  will do!  many thanks!

lurch, 2021-02-25:
@aallan I suggest that this issue and https://github.com/raspberrypi/picoprobe/issues/12 both be closed.

tge96, 2021-02-25:
Agreed.  Apologies for the noob mistake on my part.  Greatly appreciate your help!

closed 2021-02-25 by tge96