raspberrypi/pico-extras #18: stuck DMA in sd_readblocks_sync

Memotech-Bill, 2021-04-20:
After having initialised SD card access using sd_init_1pin(), calling sd_readblocks_sync() with a count > 1 results in the following error message:

stuck dma channel 11 rem 00000002 1 @ 12

The following information may or may not be relevant:

My CMakeLists.txt file includes:

    pico_enable_stdio_uart(pico-memu 1)
    pico_enable_stdio_usb(pico-memu 0)
    pico_add_extra_outputs(pico-memu)
    target_link_libraries(pico-memu PRIVATE
      pico_stdlib
      pico_sd_card
      pico_scanvideo_dpi
      pico_multicore
      tinyusb_host
      )

I am running scanvideo on core 0, and usb_host (hid) and sd_card on core 1.

My Pico is clocked at 200MHz.

I have found that after calling sd_init_1pin(), it is necessary to call setup_default_uart() again in order to obtain any further UART output.

lurch, 2021-04-20:
> I have found that after calling sd_init_1pin(), it is necessary to call setup_default_uart() again in order to obtain any further UART output.

Have you got a pin conflict, and SD and UART are trying to use the same pin?

kilograham, 2021-04-20:
> I have found that after calling sd_init_1pin(), it is necessary to call setup_default_uart() again in order to obtain any further UART output.

Sounds like your SD and UART pins are overlapping

> My Pico is clocked at 200MHz.

the prototype SD card code u r using does not deal with higher clock frequencies automatically;l you should set `PICO_SD_CARD_EXTRA_CLOCK_DIVIDER` - i recommend 4 which would give you 50Mhz which means 25Mhz SD clock which is the maximum

Memotech-Bill, 2021-04-20:
I am using the Pimoroni VGA demo board. My understanding of that is that for that the UART overlaps 4-pin SD Card, but not 1-pin.

Memotech-Bill, 2021-04-20:
As a work around of the SD card issue, I am just using a loop to call sd_readblocks_sync() multiple times with count=1, incrementing the sector number and buffer pointer after each call. That seems to be working for the moment.

robin7g, 2021-09-05:
I am getting this same stuck DMA channel error trying to use SD card on a Pimoroni VGA demo board but its inside the sd_init_1pin()

Here is my code.  

`
#include <stdio.h>
#include <ctype.h>
#include <string.h>
#include "pico/stdlib.h"

#define ENABLE_4_PIN 0
#define PICO_SD_CLK  5
#define PICO_SD_CMD  18
#define PICO_SD_DAT0_PIN 19
#define PICO_SD_DAT1_PIN 20
#define PICO_SD_DAT2_PIN 21
#define PICO_SD_DAT3_PIN 22

#include "pico/sd_card.h"

int main()
{
    set_sys_clock_khz(200000, true);
    stdio_init_all();
    sleep_ms(2000);
    printf("sdcard test starting. \n");
    
    sd_init_1pin();
    printf("sd_init_1pin done. \n");
    sd_set_clock_divider(4);
    printf("Init done. \n");
}    `

The message I get in the console is

`sdcard test starting. 
stuck dma channel 11 rem 00000002 1 @ 12`

It never completes the sd_init_1pin() call. What is odd is even when I disconnect the pico from the Pimoroni VGA demo board I get the same error. So the only conflict could be the usb connection. What am I missing?

kilograham, 2021-09-05:
I've never actually tried it with USB - there could be a problem there
you could try setting PICO_SD_CARD_EXTRA_CLOCK_DIVIDER=8 to see if that helps.
Make sure you pico_enable_stdio_uart off if the UART pins collide with the SD ones.

robin7g, 2021-09-05:
[image]

I have ruled out USB, I built the sd card circuit on a breadboard, tried different card readers etc and used UART with pins GPIO-0 and GPIO-1 just to be sure there was no conflict with the I/O lines. Also slowed the clock speed down to match the sd_card_test.c code to 48Mhz, and then switched to using exactly the same pico-extras sd_card_test.c program.  I get the same error with the sample code. 

I am still getting the same stuck DMA message in the init function. 

```
SD Card test
stuck dma channel 11 rem 00000002 1 @ 12
```

The error source is right in the middle I think of 

```
static int sd_init( bool _allow_four_data_pins)
    int sd_command(uint64_t packed_command, uint32_t *receive_buf, uint byte_length)
        rc = sd_response_dma(sd_cmd_dma_channel, SD_CMD_SM, receive_buf, byte_length, false, NULL, NULL, true, true, true);
```   

I changed `sd_cmd_dma_channel` to be 12 from 11 and it returns from sd_response_dma with a bad CRC message.  

```
.sdcard test starting. 
tsk

*** PANIC ***

bad crc %02x != %02x
```

kilograham, 2021-09-06:
There was a silly inverted flag bug in `sd_init_1pin` at one very early point - might be worth checking you have that fix (although it was from feb)

robin7g, 2021-09-06:
I have the latest SDK and Extras code from yesterday. I checked the inverting `!bytes_swap_on_read` its there in the codebase I have. I'll keep trying.

Roman-Port, 2022-05-17:
Hello! I'm having the exact same problems using the sd_card_test.c program. I can even reproduce the same crc error if I try to change the DMA channel. Did you ever figure out what was wrong and how to fix it?

Thanks!

Memotech-Bill, 2022-05-17:
No I never solved the issue with the SD card code from pico-extras.

Instead I developed my own code using SPI, implemented using PIO as the SD Card on the VGA demo board is on the wrong pins for Pico hardware SPI. You can find the low level code (sd_spi.pio and sd_spi2.c) in a couple of my repos, including https://github.com/Memotech-Bill/PicoBB/tree/master/src/pico.

Also in that repo is wrapper code which implements much of stdio.h with a FFS file-system on Pico Flash memory and FAT on the SD card, see lfswrap.h and associated files.

Roman-Port, 2022-05-18:
Thanks for the reply. I'm afraid that SPI may not be fast enough for what I'm trying to do. I'm trying to stream high sample rate audio to the SD card at 10.4 megabits/sec. Would you happen to know how fast your library can write?

Memotech-Bill, 2022-05-18:
Sorry, no I have never done a speed test.

My library is "good enough" for what I need so I have not had any need to investigate actual performance.