raspberrypi/pico-extras #20: problems compiling lwip module inside pico-extras

lealoureiro, 2021-05-12:
Hello,

Not sure if is the right place to report this issue, but i'm having problems to compile lwip lib that is inside pico-extras.

```
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/rmiieth.dir/Users/leandro/Projects/pico-extras/lib/lwip/src/core/netif.c.obj: in function `netif_set_addr':
netif.c:(.text.netif_set_addr+0x7e): undefined reference to `acd_netif_ip_addr_changed'
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: netif.c:(.text.netif_set_addr+0xb6): undefined reference to `acd_netif_ip_addr_changed'
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/rmiieth.dir/Users/leandro/Projects/pico-extras/lib/lwip/src/core/timeouts.c.obj:(.rodata.lwip_cyclic_timers+0x2c): undefined reference to `acd_tmr'
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/rmiieth.dir/Users/leandro/Projects/pico-extras/lib/lwip/src/core/ipv4/dhcp.c.obj: in function `dhcp_recv':
dhcp.c:(.text.dhcp_recv+0x416): undefined reference to `acd_start'
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/rmiieth.dir/Users/leandro/Projects/pico-extras/lib/lwip/src/core/ipv4/dhcp.c.obj: in function `dhcp_start':
dhcp.c:(.text.dhcp_start+0x42): undefined reference to `acd_add'
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: CMakeFiles/rmiieth.dir/Users/leandro/Projects/pico-extras/lib/lwip/src/core/ipv4/etharp.c.obj: in function `etharp_input':
etharp.c:(.text.etharp_input+0x72): undefined reference to `acd_arp_reply'
collect2: error: ld returned 1 exit status
make[2]: *** [rmiieth.elf] Error 1
make[1]: *** [CMakeFiles/rmiieth.dir/all] Error 2
make: *** [all] Error 2
```

The functions are defined inside **acd.c** but seems like this file is not listed for compilation and thus functions will be missing during linkage.

I'm trying to compile and test this project/repo: https://github.com/strags/pico-rmiieth , out of the box without modification.

Thanks in advance!

Regards,
Leandro

kripton, 2021-05-12:
Just out of curiosity: Does this project compile for you? https://github.com/Hermann-SW/pico-webserver

lealoureiro, 2021-05-12:
> Just out of curiosity: Does this project compile for you? https://github.com/Hermann-SW/pico-webserver

yes, this ones compiles and generates the file to upload to pico

kilograham, 2023-02-11:
lwIP is no longer in pico-extras; it is in pico-sdk as `pico_lwip`

closed 2023-02-11 by kilograham