raspberrypi/pico-extras #21: scanvideo.c fails to compile with enhanced compiler checking enabled

Memotech-Bill, 2021-06-10:
In trying to resolve a problem with one of my programs I added the command

`add_compile_options(-Werror -Wall -Wextra -Wnull-dereference)`

to my CMakeLists.txt file. This results in a number of warnings for scanvideo.c

`
/home/pi/pico/pico-extras/src/rp2_common/pico_scanvideo_dpi/scanvideo.c: In function 'scanline_id_after':
/home/pi/pico/pico-extras/src/rp2_common/pico_scanvideo_dpi/scanvideo.c:418:13: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
     if (tmp < video_mode.height - 1) {
             ^
/home/pi/pico/pico-extras/src/rp2_common/pico_scanvideo_dpi/scanvideo.c: In function 'default_scanvideo_scanline_repeat_count_fn':
/home/pi/pico/pico-extras/src/rp2_common/pico_scanvideo_dpi/scanvideo.c:1045:65: error: unused parameter 'scanline_id' [-Werror=unused-parameter]
 static uint default_scanvideo_scanline_repeat_count_fn(uint32_t scanline_id) {
                                                                 ^~~~~~~~~~~
/home/pi/pico/pico-extras/src/rp2_common/pico_scanvideo_dpi/scanvideo.c: In function 'video_24mhz_composable_adapt_for_mode':
/home/pi/pico/pico-extras/src/rp2_common/pico_scanvideo_dpi/scanvideo.c:1594:75: error: unused parameter 'program' [-Werror=unused-parameter]
 bool video_24mhz_composable_adapt_for_mode(const scanvideo_pio_program_t *program, const scanvideo_mode_t *mode,
                                                                           ^~~~~~~
/home/pi/pico/pico-extras/src/rp2_common/pico_scanvideo_dpi/scanvideo.c: In function 'video_default_adapt_for_mode':
/home/pi/pico/pico-extras/src/rp2_common/pico_scanvideo_dpi/scanvideo.c:1650:66: error: unused parameter 'program' [-Werror=unused-parameter]
 bool video_default_adapt_for_mode(const scanvideo_pio_program_t *program, const scanvideo_mode_t *mode,
                                                                  ^~~~~~~
/home/pi/pico/pico-extras/src/rp2_common/pico_scanvideo_dpi/scanvideo.c:1650:99: error: unused parameter 'mode' [-Werror=unused-parameter]
 bool video_default_adapt_for_mode(const scanvideo_pio_program_t *program, const scanvideo_mode_t *mode,
                                                                                                   ^~~~
/home/pi/pico/pico-extras/src/rp2_common/pico_scanvideo_dpi/scanvideo.c:1651:45: error: unused parameter 'modifiable_instructions' [-Werror=unused-parameter]
                                   uint16_t *modifiable_instructions) {
                                             ^~~~~~~~~~~~~~~~~~~~~~~
/home/pi/pico/pico-extras/src/rp2_common/pico_scanvideo_dpi/scanvideo.c: In function 'scanvideo_default_configure_pio':
/home/pi/pico/pico-extras/src/rp2_common/pico_scanvideo_dpi/scanvideo.c:1655:67: error: unused parameter 'offset' [-Werror=unused-parameter]
 void scanvideo_default_configure_pio(pio_hw_t *pio, uint sm, uint offset, pio_sm_config *config, bool overlay) {
                                                                   ^~~~~~
`

While I can, for now remove the -Werror option to get the build to complete, the code should ideally be free of warnings, see the discussion in https://www.raspberrypi.org/forums/viewtopic.php?f=33&t=312978

kilograham, 2021-06-10:
In the interim, you can add `-Wno-unused-parameter` as it is not the most aggregious warning.

Memotech-Bill, 2021-06-10:
It is perhaps worth noting that adding

`add_compile_options(-Wall -Wextra -Wnull-dereference)`

to pico-examples/CMakeLists.txt and compiling all the examples results in many warnings within pico-sdk. I haven't tried the same for pico-playground.

kilograham, 2021-06-10:
We check pico-sdk with

```
        -Werror
        -Wall
        -Wextra
        -Wnull-dereference
        -Wuninitialized
        -Wunused
        -Wcast-align
        -Wall
        -Wcast-qual
        -Wfloat-equal
        -Wmissing-format-attribute
        -Wconversion
        -Wsign-compare
```

If you are seeing issues with -Winline, that is a bug in the CMakeLists.txt (you can fix via `-Wno-inline` for now)

kilograham, 2021-06-10:
> It is perhaps worth noting that adding
> 
> `add_compile_options(-Wall -Wextra -Wnull-dereference)`
> 
> to pico-examples/CMakeLists.txt and compiling all the examples results in many warnings within pico-sdk. I haven't tried the same for pico-playground.

Ah, i misread what you said, I'' check that

kilograham, 2021-06-10:
it is also helpful to indicate your compiler version, since the warnings generated differ wildly between them

kilograham, 2021-06-10:
Ok i checked, and in building the examples there are some warnings in the examples themselves (and yes they should be fixed, but hasn't been gotten to yet), but none of them come from pico-sdk itself.

tinyusb however is a hot mess of warnings, which is perhaps what you are referring to (and pico-examples does build all the tinyusb examples too).