raspberrypi/pico-extras #31: Maximum Recursion Depth when adding include(pico_extras_import.cmake)

flashbtw, 2022-02-02:
Hi,
as soon as I add these two lines to my CMakeLists.txt of pico-examples:

```
set(PICO_EXTRAS_PATH "S:/Programming/Pico/Downloads/pico-extras/")
include(pico_extras_import.cmake)
```

I get an error saying 
```
PICO_SDK_PATH is S:/Programming/Pico/Downloads/pico-sdk
PICO platform is rp2040.
CMake Error at S:/Programming/Pico/Downloads/pico-sdk/pico_sdk_init.cmake:8 (if):
  Maximum recursion depth of 1000 exceeded
Call Stack (most recent call first):
  pico_sdk_import.cmake:62 (include)
  CMakeLists.txt:4 (include)

-- Configuring incomplete, errors occurred!
See also "S:/Programming/Pico/Downloads/pico-examples/build/CMakeFiles/CMakeOutput.log".
See also "S:/Programming/Pico/Downloads/pico-examples/build/CMakeFiles/CMakeError.log".
```
How can I fix it? 
Any help would be appreciated!

kilograham, 2022-02-02:
where did you add the two lines?!

flashbtw, 2022-02-02:
So I have a Download Directory with pico-examples and pico-sdk
In pico-examples there is a CMakeLists.txt, in which I added these two lines.

lurch, 2022-02-02:
What were you trying to accomplish? 
pico-extras includes pico-sdk https://github.com/raspberrypi/pico-extras/blob/master/CMakeLists.txt#L7 and pico-examples also includes pico-sdk https://github.com/raspberrypi/pico-examples/blob/master/CMakeLists.txt#L4 so it's not surprising that something went wrong somewhere.

kilograham, 2022-02-03:
i meant where in the CMakeLists.txt did you add the lines?

flashbtw, 2022-02-03:
> pico-extras includes pico-sdk https://github.com/raspberrypi/pico-extras/blob/master/CMakeLists.txt#L7
I just copied the external/pico_extras_import.cmake and copied them into a file on my computer.
Sorry, I didn't explain it well enough. 

> i meant where in the CMakeLists.txt did you add the lines?
below project(...)

I tried it everywhere, everytime the same error. I can send you a zip of my files and their structure, if you want.

flashbtw, 2022-02-03:
so, for a better recreation of the problem, just clone the pico-playground repository on Windows 10, `cd pico-playground` and `mkdir build` in it and `cd build`. Then with the developer PowerShell of Visual Studio 2022 run `cmake -G "NMake Makefiles .."` and you'll get the same error as me. (Console Output is: https://imgur.com/a/wJflLUC)

My File Structure looks like this:
```
S:\
  Other Stuff
  Programming
    Other Stuff
    Pico
      GNU Arm
      CMake
      Downloads
        pico-sdk (got with git clone -b master https://github.com/raspberrypi/pico-sdk)
        pico-playground (got with git clone -b master https://github.com/raspberrypi/pico-playground)
          build (Run cmake "NMake Makefiles" .. here)
```

Thank you!