raspberrypi/pico-extras #42: [Help Wanted] Why pico_scanvideo_dpi take both UART pins?

lightsing, 2022-08-22:
I need to produce VGA from UART input (yeah, just a terminal emulator).
However pico_scanvideo_dpi took both UART0 and UART1 pins.
Can I relocate the output pins of the vga signal to other pins?

Thanks.

lurch, 2022-08-22:
I dunno if you managed to solve your problem, but you can also move the UARTs onto different pins, which is probably easier than trying to move the VGA pins?
If you look at section 1.4.3. of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf you can see that UART1 TX and RX are available on GPIO 20 and 21. However these aren't shown on https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#pinout-and-design-files because the RTS pin in that block of UART-GPIOs (20-23) isn't available (externally) on the Pico.

lightsing, 2022-08-23:
> However these aren't shown on https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#pinout-and-design-files because the RTS pin in that block of UART-GPIOs (20-23) isn't available (externally) on the Pico.

But it's possible to wire those two from pico board?

Currently, I reassigned RGB pins to GP6-GP20, VSYNC to GP26 and HSYNC to GP27. I'm not sure if this combination works (waiting for the PCB to test out).
[image]

kilograham, 2022-08-23:
You could move the VGA output, but much easier to just use UART1 on pins 20/21

e.g. using `PICO_BOARD=vgaboard`

lightsing, 2022-09-05:
@lurch sorry for the bother. I tried the offical layout and it works pretty well.
[image]

and I also tried the following layout which assign :
- R0-R4 to GP6-10
- alpha pin to GP11
- G0-G4 to GP12-GP16
- B0-B4 to GP18-21
- HSYNC at GP26 and VSYNC at GP27
[image]

I also change the corresponding header, but it just not working and UART0 also not working(TX using GP0):
```c
#undef PICO_SCANVIDEO_COLOR_PIN_COUNT
#define PICO_SCANVIDEO_COLOR_PIN_COUNT 16
#undef PICO_SCANVIDEO_PIXEL_BCOUNT
#define PICO_SCANVIDEO_PIXEL_BCOUNT 5
#undef PICO_SCANVIDEO_PIXEL_GCOUNT
#define PICO_SCANVIDEO_PIXEL_GCOUNT 5
#undef PICO_SCANVIDEO_PIXEL_RCOUNT
#define PICO_SCANVIDEO_PIXEL_RCOUNT 5

#undef PICO_SCANVIDEO_COLOR_PIN_BASE
#define PICO_SCANVIDEO_COLOR_PIN_BASE 6
#undef PICO_SCANVIDEO_ALPHA_PIN
#define PICO_SCANVIDEO_ALPHA_PIN 11
// R uses GP6-10
#undef PICO_SCANVIDEO_PIXEL_RSHIFT
#define PICO_SCANVIDEO_PIXEL_RSHIFT 0u
// G uses GP12-16
#undef PICO_SCANVIDEO_PIXEL_GSHIFT
#define PICO_SCANVIDEO_PIXEL_GSHIFT 6u
// B uses GP17-21
#undef PICO_SCANVIDEO_PIXEL_BSHIFT
#define PICO_SCANVIDEO_PIXEL_BSHIFT 11u

// HSYNC -> GP26
// VSYNC -> GP27
#undef PICO_SCANVIDEO_SYNC_PIN_BASE
#define PICO_SCANVIDEO_SYNC_PIN_BASE 26
```

lightsing, 2022-09-05:
thx!

> I'm afraid I have no idea, I've never actually run any of these VGA examples myself. @kilograham would be the person to ask, but he's always _very_ busy, so you might have more luck asking for help at https://forums.raspberrypi.com/viewforum.php?f=143

closed 2023-06-13 by kilograham