raspberrypi/pico-extras #43: Remove LWIP from pico-extras ?

lurch, 2022-09-21:
Now that lwIP has been integrated into [pico-sdk](https://github.com/raspberrypi/pico-sdk/tree/master/src/rp2_common/pico_lwip) itself (as part of the Pico W functionality), _perhaps_ it makes sense to remove it from [pico-extras](https://github.com/raspberrypi/pico-extras/tree/master/src/rp2_common/lwip) ? And for the lwIP-related issues and PRs in this repo to either be closed, or transferred to pico-sdk, as appropriate?
#19 
#20 
#27 
#33 
#7 
#35

lurch, 2023-02-11:
Fixed by https://github.com/raspberrypi/pico-extras/commit/5e594d417d10adc341eb9a884810022f2f6ca5cc

@kilograham You might want to go through and close all the other lwIP-related issues in my comment above?

closed 2023-02-11 by lurch

kilograham, 2023-02-11:
ah thanks, i had closed some of those but not all!