raspberrypi/pico-extras #54: Just researching this and have small question. Is it possible to use an external clock provider ie Si5338  ?

kenkit, 2022-12-28:
Hi, 
I'm wondering if it's possible to use an external clock provider chip like Si5338 to provide clock signal.
Since the above chip is programmable we could have it provide any frequency ?

lurch, 2022-12-28:
It's not something I've done myself, but section 2.16 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf says
"If the user already has an accurate clock source then it is possible to drive an external clock directly into XIN (aka XI), and disable the oscillator circuit. In this mode XIN can be driven at up to 50MHz."

lurch, 2022-12-28:
You might also find https://www.raspberrypi.com/news/dont-try-this-at-home-overclocking-rp2040-to-1ghz/ interesting / useful.