raspberrypi/pico-extras #63: Schematic or hardware specification for scanvideo.c ?

Andy2No, 2023-04-24:
I see the VGA generator is based on 5:5:5 RGB pixels, but I don't see any details of the video DAC used, or a schematic for one.  I assume a resistor based DAC is intended?  If so, what resistor values are suitable?  

Do the hsync and vsync signals also need resistors?  Is there any other circuitry needed besides just resistors?

A simple schematic would be useful.  Failing that, just a clear description of how to build it.

On first look, I don't even see how the GPIO pins are assigned, or any obvious documentation about that.  I'd presumably need to wade through the code to try to figure it out.  A simple table of pin assignments would be a good start.

JamesH65, 2023-04-24:
I believe the schematic on this page is the relevant one? https://github.com/raspberrypi/pico-playground

Andy2No, 2023-04-24:
Thanks @JamesH65 

It would be good if someone could add that link to the documentation, maybe in the readme files at:
 https://github.com/raspberrypi/pico-extras/tree/master/src/common/pico_scanvideo

and / or https://github.com/raspberrypi/pico-extras

JamesH65, 2023-04-24:
I've made some changes, the PR is linked just above. Would that be adequate?

lurch, 2023-04-24:
> I believe the schematic on this page is the relevant one? https://github.com/raspberrypi/pico-playground 

See also https://github.com/raspberrypi/pico-playground/issues/20 :confused:

Andy2No, 2023-04-25:
Thanks, @JamesH65 and @lurch 

Yes, the PDF seems like the best thing to link to, and just linking to it from the https://github.com/raspberrypi/pico-extras readme is probably enough.  That's one of the places I looked, when I was trying to figure it out.

I'd be inclined to add it to the readme for each part of pico-extras where it's relevant too, and maybe add a comment to the main .c files, but maybe that's overkill.

JamesH65, 2023-04-28:
Change ha been merged so closing.

JamesH65, 2023-04-28:
@kilograham Don't appear to be able to close issues on this tracker, but this one is OK to close.

closed 2023-04-28 by aallan