raspberrypi/pico-extras #71: [Help Wanted] examples or success stories for scanvideo with parallel LCD (RGB, H+VSYNC, DEN + PCLK)

PR77, 2024-01-09:
Has anyone had success with using scanvideo with a parallel LCD (i.e., 8 Bits RED, GREEN + BLUE, H+VSYNC, DEN and PCLK)? I have this working with the Pi B+ (https://forums.raspberrypi.com/viewtopic.php?t=86658&start=150#p691260) using DPI, but would really like to try on the Pico.

I haven't yet tried myself but will do soon. Additionally, any idea where the GPIO pins are defined in the scanvideo library? I haven't really found where DEN and PCLK could be defined if PICO_SCANVIDEO_ENABLE_DEN_PIN and PICO_SCANVIDEO_ENABLE_CLOCK_PIN are defined.

lurch, 2024-01-09:
> I haven't yet tried myself but will do soon.

In which case you should probably be asking for help at https://forums.raspberrypi.com/ rather than GitHub. We tend to use GitHub for tracking bugs, not for help-requests.

PR77, 2024-01-15:
Existing thread found with some additional details - https://forums.raspberrypi.com/viewtopic.php?p=2178610#p2178610.

closed 2024-01-15 by PR77