raspberrypi/pico-extras #82: Where is scanvideo.pio.h ? Unable to build pico_scanvideo demo1

Andy2No, 2024-08-25:
src/common/pico_scanvideo/include/pico/scanvideo/composable_scanline.h

tries to include scanvideo.pio.h - which as far as I can tell, does not exist.

I've tried searching the organisation with path:scanvideo.pio.h but found nothing.

FWIW, I'm trying to import demo1 from scanvideo in pico-playground into the Arduino IDE, by pasting in whatever seems to be needed, but that file appears to be missing from any of the repos.

lurch, 2024-08-25:
`scanvideo.pio.h` gets [automatically generated](https://github.com/raspberrypi/pico-extras/blob/master/src/common/pico_scanvideo/CMakeLists.txt#L6) from [`scanvideo.pio`](https://github.com/raspberrypi/pico-extras/blob/master/src/common/pico_scanvideo/scanvideo.pio).

> ... the Arduino IDE

I have no familiarity with the Arduino IDE, but maybe you need to do a least one build with `cmake`, and that should generate the `scanvideo.pio.h` that you need.

Andy2No, 2024-08-26:
Okay, yes.  Thanks.

closed 2024-08-26 by Andy2No