raspberrypi/pico-extras #85: Instructions should be updated

alidaf, 2024-12-04:
The instructions for adding the pico-extras library to a project don't work for VS Code and should be updated to reflect VS Code support in the "Getting Started Guide".

lurch, 2024-12-04:
ping @will-v-pi and @nathan-contino

will-v-pi, 2024-12-05:
The instructions do work for VS Code for me, so I don't think there's any docs updates needed

lurch, 2024-12-05:
@alidaf Can you please provide more details of exactly what you did, and how it "doesn't work" ?

alidaf, 2024-12-06:
Do you mean `target_link_libraries` as I don't see a `pico_link_libraries`?

will-v-pi, 2024-12-06:
Yes, I meant `target_link_libraries`

alidaf, 2024-12-06:
I do now but it makes no difference. Even the root `pico-extras` in the include is not recognised.

will-v-pi, 2024-12-06:
Ah, it should be `pico/audio_i2s.h`, not `pico-extras/audio_i2s.h` - see the [pico-playground audio example](https://github.com/raspberrypi/pico-playground/tree/master/audio)

alidaf, 2024-12-06:
It's still not recognised. The CMakeLists.txt in that example does not look like the autogenerated one in which I've updated the `target_link_libraries`.

alidaf, 2024-12-06:
Actually, the wiggly line disappears after a compile. I think I'm good to go.

alidaf, 2024-12-06:
Yes, that seems to be it. Many thanks. Is the `target_link_libraries` something that can go in the instructions?

closed 2024-12-06 by alidaf

will-v-pi, 2024-12-06:
It's already documented in section 2.4 of the [C/C++ SDK PDF](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf) - adding dependencies to `target_link_libraries` is required for all pico-sdk and pico-extras libraries