raspberrypi/pico-extras #88: Non-english letters - Compile issue

JDM-Danisense, 2025-01-15:
Hi

I have tried with a simple example project to compile it - but without luck. The error code looks like this: 

```
ninja: Entering directory `C:\Code\spi_dma/build'
[1/1] Linking CXX executable spi_dma.elf
FAILED: spi_dma.elf 
C:\Windows\system32\cmd.exe /C "cd . && C:\Users\JacobD├╕llnerM├╕nster\.pico-sdk\toolchain\13_3_Rel1\bin\arm-none-eabi-g++.exe -mcpu=cortex-m0plus -mthumb -g -O3 -DNDEBUG -Wl,--build-id=none -Wl,-Map=spi_dma.elf.map --specs=nosys.specs -Wl,--wrap=__clzsi2 -Wl,--wrap=__clzdi2 -Wl,--wrap=__ctzsi2 -Wl,--wrap=__popcountsi2 -Wl,--wrap=__popcountdi2 -Wl,--wrap=__clz -Wl,--wrap=__clzl -Wl,--wrap=__clzll -Wl,--wrap=__ctzdi2 -Wl,--wrap=__aeabi_idiv -Wl,--wrap=__aeabi_idivmod -Wl,--wrap=__aeabi_ldivmod -Wl,--wrap=__aeabi_uidiv -Wl,--wrap=__aeabi_uidivmod -Wl,--wrap=__aeabi_uldivmod -Wl,--wrap=__aeabi_dadd -Wl,--wrap=__aeabi_ddiv -Wl,--wrap=__aeabi_dmul -Wl,--wrap=__aeabi_drsub -Wl,--wrap=__aeabi_dsub -Wl,--wrap=__aeabi_cdcmpeq -Wl,--wrap=__aeabi_cdrcmple -Wl,--wrap=__aeabi_cdcmple -Wl,--wrap=__aeabi_dcmpeq -Wl,--wrap=__aeabi_dcmplt -Wl,--wrap=__aeabi_dcmple -Wl,--wrap=__aeabi_dcmpge -Wl,--wrap=__aeabi_dcmpgt -Wl,--wrap=__aeabi_dcmpun -Wl,--wrap=__aeabi_i2d -Wl,--wrap=__aeabi_l2d -Wl,--wrap=__aeabi_ui2d -Wl,--wrap=__aeabi_ul2d -Wl,--wrap=__aeabi_d2iz -Wl,--wrap=__aeabi_d2lz -Wl,--wrap=__aeabi_d2uiz -Wl,--wrap=__aeabi_d2ulz -Wl,--wrap=__aeabi_d2f -Wl,--wrap=sqrt -Wl,--wrap=cos -Wl,--wrap=sin -Wl,--wrap=tan -Wl,--wrap=atan2 -Wl,--wrap=exp -Wl,--wrap=log -Wl,--wrap=ldexp -Wl,--wrap=copysign -Wl,--wrap=trunc -Wl,--wrap=floor -Wl,--wrap=ceil -Wl,--wrap=round -Wl,--wrap=sincos -Wl,--wrap=asin -Wl,--wrap=acos -Wl,--wrap=atan -Wl,--wrap=sinh -Wl,--wrap=cosh -Wl,--wrap=tanh -Wl,--wrap=asinh -Wl,--wrap=acosh -Wl,--wrap=atanh -Wl,--wrap=exp2 -Wl,--wrap=log2 -Wl,--wrap=exp10 -Wl,--wrap=log10 -Wl,--wrap=pow -Wl,--wrap=powint -Wl,--wrap=hypot -Wl,--wrap=cbrt -Wl,--wrap=fmod -Wl,--wrap=drem -Wl,--wrap=remainder -Wl,--wrap=remquo -Wl,--wrap=expm1 -Wl,--wrap=log1p -Wl,--wrap=fma -Wl,--wrap=__aeabi_lmul -Wl,--wrap=__aeabi_fadd -Wl,--wrap=__aeabi_fdiv -Wl,--wrap=__aeabi_fmul -Wl,--wrap=__aeabi_frsub -Wl,--wrap=__aeabi_fsub -Wl,--wrap=__aeabi_cfcmpeq -Wl,--wrap=__aeabi_cfrcmple -Wl,--wrap=__aeabi_cfcmple -Wl,--wrap=__aeabi_fcmpeq -Wl,--wrap=__aeabi_fcmplt -Wl,--wrap=__aeabi_fcmple -Wl,--wrap=__aeabi_fcmpge -Wl,--wrap=__aeabi_fcmpgt -Wl,--wrap=__aeabi_fcmpun -Wl,--wrap=__aeabi_i2f -Wl,--wrap=__aeabi_l2f -Wl,--wrap=__aeabi_ui2f -Wl,--wrap=__aeabi_ul2f -Wl,--wrap=__aeabi_f2iz -Wl,--wrap=__aeabi_f2lz -Wl,--wrap=__aeabi_f2uiz -Wl,--wrap=__aeabi_f2ulz -Wl,--wrap=__aeabi_f2d -Wl,--wrap=sqrtf -Wl,--wrap=cosf -Wl,--wrap=sinf -Wl,--wrap=tanf -Wl,--wrap=atan2f -Wl,--wrap=expf -Wl,--wrap=logf -Wl,--wrap=ldexpf -Wl,--wrap=copysignf -Wl,--wrap=truncf -Wl,--wrap=floorf -Wl,--wrap=ceilf -Wl,--wrap=roundf -Wl,--wrap=sincosf -Wl,--wrap=asinf -Wl,--wrap=acosf -Wl,--wrap=atanf -Wl,--wrap=sinhf -Wl,--wrap=coshf -Wl,--wrap=tanhf -Wl,--wrap=asinhf -Wl,--wrap=acoshf -Wl,--wrap=atanhf -Wl,--wrap=exp2f -Wl,--wrap=log2f -Wl,--wrap=exp10f -Wl,--wrap=log10f -Wl,--wrap=powf -Wl,--wrap=powintf -Wl,--wrap=hypotf -Wl,--wrap=cbrtf -Wl,--wrap=fmodf -Wl,--wrap=dremf -Wl,--wrap=remainderf -Wl,--wrap=remquof -Wl,--wrap=expm1f -Wl,--wrap=log1pf -Wl,--wrap=fmaf -Wl,--wrap=malloc -Wl,--wrap=calloc -Wl,--wrap=realloc -Wl,--wrap=free -Wl,--wrap=memcpy -Wl,--wrap=memset -Wl,--wrap=__aeabi_memcpy -Wl,--wrap=__aeabi_memset -Wl,--wrap=__aeabi_memcpy4 -Wl,--wrap=__aeabi_memset4 -Wl,--wrap=__aeabi_memcpy8 -Wl,--wrap=__aeabi_memset8 -Wl,-LC:/Code/spi_dma/build -Wl,--script=C:/Users/JacobD├╕llnerM├╕nster/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_crt0/rp2040/memmap_default.ld -Wl,-z,max-page-size=4096 -Wl,--gc-sections -Wl,--wrap=sprintf -Wl,--wrap=snprintf -Wl,--wrap=vsnprintf -Wl,--wrap=printf -Wl,--wrap=vprintf -Wl,--wrap=puts -Wl,--wrap=putchar -Wl,--wrap=getchar @CMakeFiles\spi_dma.rsp -o spi_dma.elf && C:\Windows\system32\cmd.exe /C "cd /D C:\Code\spi_dma\build && C:\Users\JacobD├╕llnerM├╕nster\.pico-sdk\toolchain\13_3_Rel1\bin\arm-none-eabi-objdump.exe -h C:/Code/spi_dma/build/spi_dma.elf > spi_dma.dis && C:\Users\JacobD├╕llnerM├╕nster\.pico-sdk\toolchain\13_3_Rel1\bin\arm-none-eabi-objdump.exe -d C:/Code/spi_dma/build/spi_dma.elf >> spi_dma.dis && cd /D C:\Code\spi_dma\build && C:\Users\JacobD├╕llnerM├╕nster\.pico-sdk\toolchain\13_3_Rel1\bin\arm-none-eabi-objcopy.exe -Oihex C:/Code/spi_dma/build/spi_dma.elf spi_dma.hex && cd /D C:\Code\spi_dma\build && C:\Users\JacobD├╕llnerM├╕nster\.pico-sdk\toolchain\13_3_Rel1\bin\arm-none-eabi-objcopy.exe -Obinary C:/Code/spi_dma/build/spi_dma.elf spi_dma.bin && cd /D C:\Code\spi_dma\build && C:\Users\JacobD├╕llnerM├╕nster\.pico-sdk\picotool\2.1.0\picotool\picotool.exe uf2 convert --quiet C:/Code/spi_dma/build/spi_dma.elf spi_dma.uf2 --family rp2040""
C:/Users/JacobD°llnerM°nster/.pico-sdk/toolchain/13_3_Rel1/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file C:/Users/JacobD├╕llnerM├╕nster/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_crt0/rp2040/memmap_default.ld: Invalid argument
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Users\JacobDøllnerMønster\.pico-sdk\ninja\v1.12.1\ninja.exe '-C', 'C:\Code\spi_dma/build'" terminated with exit code: 1. 
```

My guess that the reason for this issue is the inclusion of the danish letter "ø". 

Possible solution: make the download location of the SDK be settable by the user. 

System: Win10
Language: Danish

lurch, 2025-01-15:
I'm guessing that you're using https://github.com/raspberrypi/pico-vscode ?
ping @will-v-pi

will-v-pi, 2025-01-15:
Yes, there is a known issue with the VS Code extension, that it currently has issues with non-english language settings - https://github.com/raspberrypi/pico-vscode/issues/136

Although this particular issue is new - the usual issue is just that it fails to create a new project. This one looks related to the CMake/Ninja interaction instead, as it looks like the compiler is being passed an incorrect file path from Ninja.

JDM-Danisense, 2025-01-15:
> I'm guessing that you're using https://github.com/raspberrypi/pico-vscode ? ping @will-v-pi

Indeed - Sorry, I can see there that I placed my issue a completely wrong repo. 
Thanks for helping!

@will-v-pi Should i open a new/move this issue to the correct repo?