raspberrypi/pico-extras #89: Missing import in sleep.h?

a-mueller, 2025-03-03:
I am pretty new to C so this might be stupid.

I am using a RP2040 and I am trying to go dormant. When using `sleep_run_from_xosc` it complains that `hardware_alarm_callback_t` is not defined in this scope. Adding `#include "hardware/timer.h"` to `pico-extras/src/rp2_common/pico_sleep/include/pico/sleep.h` will fix it but I am pretty sure some kind of test would have caught that the whole thing doesn't compile. Anything I am doing wrong here?

Thanks

lurch, 2025-03-04:
ping @peterharperuk as he was the last person to edit the `pico_sleep` code 😴  (in #79 ).

> I am pretty sure some kind of test would have caught that the whole thing doesn't compile.

The code in `pico-extras` (and `pico-playground`) is unsupported / experimental, and so it gets **much** less testing than the code in `pico-sdk` and `pico-examples`.

peterharperuk, 2025-03-05:
Yes, good catch. The test code builds fine but it has #include "pico/stdlib.h" before #include "pico/sleep.h" which includes "hardware/timer.h" for us. As a workaround just include it before including sleep.h and I'll push a fix.

PR #90 (2025-08-12): Include hardware/timer.h
If you include sleep.h first you get a compile error as hardware_alarm_callback_t is not defined. Fix this by including the appropriate header.

Fixes #89
commit eb071cd8 (2025-08-12): Include hardware/timer.h (#90)

If you include sleep.h first you get a compile error as
hardware_alarm_callback_t is not defined. Fix this by including the
appropriate header.

Fixes #89