raspberrypi/pico-extras #96: Scanvideo DPI doesn't support RP2350B variant with pin base > 29.

peterk268, 2025-08-01:
When using the scanvide dpi library. I notice that the same code I used on the RP2350A variant works but not on the RP2350B variant with a sync base of 41. I noticed in the code there are the pin masks which use 32 bit values on line 1378 in scanvideo.c 
Now I also tried to manually set my pins as pio pins but that didn't work either so I believe there are underlying issues with this since this was initially written for the RP2040 and probably not updated for the RP2350B variant with 48 GPIO instead of the initial 30.

```
    uint pin_mask = 3u << PICO_SCANVIDEO_SYNC_PIN_BASE;
    bi_decl_if_func_used(bi_2pins_with_names(PICO_SCANVIDEO_SYNC_PIN_BASE, "HSync",
                                               PICO_SCANVIDEO_SYNC_PIN_BASE + 1, "VSync"));

#if PICO_SCANVIDEO_ENABLE_DEN_PIN
    bi_decl_if_func_used(bi_1pin_with_name(PICO_SCANVIDEO_SYNC_PIN_BASE + 2, "Display Enable"));
    pin_mask |= 4u << PICO_SCANVIDEO_SYNC_PIN_BASE;
#endif
#if PICO_SCANVIDEO_ENABLE_CLOCK_PIN
    bi_decl_if_func_used(bi_1pin_with_name(PICO_SCANVIDEO_SYNC_PIN_BASE + 3, "Pixel Clock"));
    pin_mask |= 8u << PICO_SCANVIDEO_SYNC_PIN_BASE;
#endif
    static_assert(PICO_SCANVIDEO_PIXEL_RSHIFT + PICO_SCANVIDEO_PIXEL_RCOUNT <= PICO_SCANVIDEO_COLOR_PIN_COUNT, "red bits do not fit in color pins");
    static_assert(PICO_SCANVIDEO_PIXEL_GSHIFT + PICO_SCANVIDEO_PIXEL_GCOUNT <= PICO_SCANVIDEO_COLOR_PIN_COUNT, "green bits do not fit in color pins");
    static_assert(PICO_SCANVIDEO_PIXEL_BSHIFT + PICO_SCANVIDEO_PIXEL_BCOUNT <= PICO_SCANVIDEO_COLOR_PIN_COUNT, "blue bits do not fit in color pins");
#define RMASK ((1u << PICO_SCANVIDEO_PIXEL_RCOUNT) - 1u)
#define GMASK ((1u << PICO_SCANVIDEO_PIXEL_GCOUNT) - 1u)
#define BMASK ((1u << PICO_SCANVIDEO_PIXEL_BCOUNT) - 1u)
    pin_mask |= RMASK << (PICO_SCANVIDEO_COLOR_PIN_BASE + PICO_SCANVIDEO_PIXEL_RSHIFT);
    pin_mask |= GMASK << (PICO_SCANVIDEO_COLOR_PIN_BASE + PICO_SCANVIDEO_PIXEL_GSHIFT);
    pin_mask |= BMASK << (PICO_SCANVIDEO_COLOR_PIN_BASE + PICO_SCANVIDEO_PIXEL_BSHIFT);
    bi_decl_if_func_used(bi_pin_mask_with_name(RMASK << (PICO_SCANVIDEO_COLOR_PIN_BASE + PICO_SCANVIDEO_PIXEL_RSHIFT), RMASK == 1 ? "Red" : ("Red 0-" __XSTRING(PICO_SCANVIDEO_PIXEL_GCOUNT))));
    bi_decl_if_func_used(bi_pin_mask_with_name(GMASK << (PICO_SCANVIDEO_COLOR_PIN_BASE + PICO_SCANVIDEO_PIXEL_GSHIFT), GMASK == 1 ? "Green" : ("Green 0-" __XSTRING(PICO_SCANVIDEO_PIXEL_GCOUNT))));
    bi_decl_if_func_used(bi_pin_mask_with_name(BMASK << (PICO_SCANVIDEO_COLOR_PIN_BASE + PICO_SCANVIDEO_PIXEL_BSHIFT), BMASK == 1 ? "Blue" : ("Blue 0-" __XSTRING(PICO_SCANVIDEO_PIXEL_BCOUNT))));

    for(uint8_t i = 0; pin_mask; i++, pin_mask>>=1u) {
        if (pin_mask & 1) gpio_set_function(i, GPIO_FUNC_PIO0);
    }
```

peterk268, 2025-08-07:
Hi, I've managed to solve the issue. The issue comes from both the scanvideo library and the pio library so I would have to do 2 pull requests. I'll add links to both pull requests here in case someone stumbles upon this.

lurch, 2025-08-07:
If you'd like to suggest a change, you'll need to create a [Pull Request](https://github.com/raspberrypi/pico-extras/pulls), rather than attaching a zipfile of source-code. Thanks.

(But note that pico-extras and pico-playground are "experimental / unsupported, use at your own risk", and so don't see much active development)